voodootikigod/node-rolling-spider #11: camera support?

sidorares, 2015-05-10:
would be nice to have api to access a camera. Not sure if firmware supports live data transferr ( still or video) but at least "grab a picture and save it" api. Happy to submit a code if anyone managed to document / dump camera commands.

voodootikigod, 2015-05-11:
This is actually next on my punchlist.

calweb, 2015-05-11:
awesome! :+1:

wijnanb, 2015-05-19:
+1
Would love to see this.

bpyser, 2015-05-29:
Working on this right now with @johnelliott at JSConf2015!

trezy, 2015-06-05:
Any luck @bpyser or @johnelliott? I was working on this as well as swarm at JSConf.

voodootikigod, 2015-06-06:
I have swarm done, polishing it up right now..

Camera is next on my punchlist.

*Chris Williams*

@voodootikigod <http://twitter.com/voodootikigod> | GitHub
<http://github.com/voodootikigod>

*The things I make that you should check out:*
SaferAging <http://www.saferaging.com/> | JSConf <http://jsconf.com/> |
RobotsConf <http://robotsconf.com/> | RobotsWeekly
<http://robotsweekly.com/>

Help me end the negativity on the internet, share this
<http://jsconf.eu/2011/an_end_to_negativity.html>.

johnelliott, 2015-06-06:
Ben was looking at it today. I think he has taken images, but we haven't gotten any off. He was using a bluetooth logger on android to capture what was happening when photos are taken using the FreeFlight3 phone app.

I am playing with the idea of an in-browser ground control station to take in the images if we can get them coming in over bluetooth. This may be a good way to control swarms too.

rashedtalukder, 2015-10-12:
Just wondering if any progress has been made on this feature.

johnelliott, 2015-10-12:
@rashedtalukder I haven't heard anything. There seems to be more activity overall on the manufacturer site: http://forum.developer.parrot.com/c/drone-sdk/rolling-spider but not specifically related. We haven't really done anything with the rolling spider at Social Tables for a bit :/

rashedtalukder, 2015-10-20:
@johnelliott that's too bad. I replied in this thread asking for more details from Parrot on making the call to capture and retrieve the image from the device, but I haven't gotten a reply from them. I'd love it if this could be made to work =/

http://forum.developer.parrot.com/t/bluetooth-attribute-definitions-for-roll-pitch-yaw-rotation-and-altitude/105/4

jjant, 2017-05-19:
Any news on this?