voodootikigod/node-rolling-spider #25: File not found for discovery tool

slooker, 2015-05-27:
The discovery tool listed in the README.md is not found.  

https://github.com/FluffyJack/node-rolling-spider/blob/master/SamplesAndTools/discover.js

twilson63, 2015-05-27:
check out the `eg` folder

-- 
Tom Wilson
Jack Russell Software, A Tabula Rasa Healthcare Division
494 Wando Park Blvd
Mount Pleasant, SC  29464
Phone: 843-606-6484
Mobile: 843-469-5856
Email: tom@jackhq.com
Web: http://www.jackhq.com
Calendar:
http://www.google.com/calendar/embed?src=tom%40jackrussellsoftware.com&ctz=America/New_York
<http://www.jackhq.com/calendar>

This e-mail may contain information that is confidential, privileged or
otherwise protected from disclosure by the Health Insurance Portability and
Accountability Act (HIPAA) and other state and federal laws.  This
information is intended only for the individual names above.  Any review,
use disclosure or dissemination of this material is strictly prohibited.
If you receive this information in error, please notify CareKinesis
immediately at 888-974-2763 and delete the original at once.

slooker, 2015-05-27:
I saw... just wanted to point out that the url was wrong in the README :)

voodootikigod, 2015-06-02:
This has been fixed.

closed 2015-06-02 by voodootikigod