voodootikigod/node-rolling-spider #27: Windows Support ?!

moszeed, 2015-05-29:
Hi i cant join the fun to fly my Rolling-Spider with Node,
because i can`t install this module on my Windows 8 or 10 Machine.
Following message appears:
```
λ npm install rolling-spider
npm ERR! Windows_NT 6.3.9600
npm ERR! argv "C:\\Program Files\\nodejs\\\\node.exe" "C:\\Program Files\\nodejs\\node_modules \\npm\\bin\\npm-cli.js" "install" "rolling-spider"
npm ERR! node v0.12.4
npm ERR! npm  v2.10.1
npm ERR! code EBADPLATFORM

npm ERR! notsup Unsupported
npm ERR! notsup Not compatible with your operating system or architecture: rolling-spider@1.1.2
npm ERR! notsup Valid OS:    darwin,linux
npm ERR! notsup Valid Arch:  any
npm ERR! notsup Actual OS:   win32
npm ERR! notsup Actual Arch: x64
```
Thanks in advance

ChrisTheBaron, 2015-05-29:
Hey, 

This is because the Bluetooth module it depends on only runs on Linux and iOS at the moment, sorry :) 

Cheers, Chris

blhack, 2015-05-29:
I think ChrisTheBaron means Linux and OSX, right?  (This does run on OSX [yosemite, at least.  Having trouble with mavericks right now]).

ChrisTheBaron, 2015-05-29:
Hey, 

Yes that is what I meant, thank you :) 

Cheers, Chris

moszeed, 2015-06-01:
hm, thats sad :(

voodootikigod, 2015-06-01:
Yes it is sad I am trying to work with @microsoft to get their ble Apis not
to be a nightmare

voodootikigod, 2015-06-02:
@blhack can you provide logs from mavericks?

sandeepmistry, 2015-06-02:
@voodootikigod apologies for going off topics ...

@blhack's going to try a patch out for me. It looks like OS X 10.9 stopped emitting an event for write without response requests, so the write callback is never called, and things get stuck. If all goes well the latest noble will have a fix for this.

voodootikigod, 2015-06-08:
Closing as the root issue is with Noble and windows support (which has a ticket already).

closed 2015-06-08 by voodootikigod