voodootikigod/node-rolling-spider #38: Cannot call method 'discoverAllServicesAndCharacteristics' of null

Resseguie, 2015-06-03:
OS X 10.9.5
XCode Version 6.2 (6C131e)

Code: (`ping.js`)
``` js
var RollingSpider = require("rolling-spider");
var myDrone = new RollingSpider();

myDrone.connect(function() {
  myDrone.setup(function() {
    myDrone.startPing();
  });
});
```

Full error: (`node ping`)
```bash
/Users/Resseguie/Documents/drone/node_modules/rolling-spider/lib/drone.js:216
  this.peripheral.discoverAllServicesAndCharacteristics(function (error, servi
                  ^
TypeError: Cannot call method 'discoverAllServicesAndCharacteristics' of null
    at Drone.setup (/Users/Resseguie/Documents/drone/node_modules/rolling-spider/lib/drone.js:216:19)
    at /Users/Resseguie/Documents/drone/ping.js:5:11
    at null.<anonymous> (/Users/Resseguie/Documents/drone/node_modules/rolling-spider/lib/drone.js:188:11)
    at Noble.emit (events.js:95:17)
    at Noble.onStateChange (/Users/Resseguie/Documents/drone/node_modules/rolling-spider/node_modules/noble/lib/noble.js:66:8)
    at emit (events.js:95:17)
    at nobleBindings.startScanning.args.kCBMsgArgOptions (/Users/Resseguie/Documents/drone/node_modules/rolling-spider/node_modules/noble/lib/mac/mavericks.js:73:8)
    at emit (events.js:95:17)
    at nobleBindings.sendCBMsg.sendXpcMessage.kCBMsgId (/Users/Resseguie/Documents/drone/node_modules/rolling-spider/node_modules/noble/lib/mac/mavericks.js:43:8)
    at emit (events.js:95:17)
```

I tried with both the npm version of rolling-spider and the latest from the `voodootikigod/node-rolling-spider` repo. No errors on `npm install`.

Is the latest version of XCode (and thus Yosemite) required?

sandeepmistry, 2015-06-05:
@Resseguie can you please provide debug output?

I'm able to reproduce on Yosemite, if I turn Bluetooth off:

```sh
Sandeep-Mistrys-MacBook-Air:rolling-spider smistry$ DEBUG=* node issue38.js 
  yosemite-bindings sendCBMsg: 1, {
  "kCBMsgArgName": "node-1433466289629",
  "kCBMsgArgOptions": {
    "kCBInitOptionShowPowerAlert": 0
  },
  "kCBMsgArgType": 0
} +0ms
  rollingspider RollingSpider#connect +0ms
  rollingspider RollingSpider.on(stateChange) +2ms
  yosemite-bindings xpcEvent: {
  "kCBMsgId": 6,
  "kCBMsgArgs": {
    "kCBMsgArgState": 4
  }
} +72ms
  yosemite-bindings state change poweredOff +0ms
  noble stateChange poweredOff +0ms
  rollingspider stateChange == poweredOff +3ms
  yosemite-bindings sendCBMsg: 30, null +0ms
  noble scanStop +1ms
  rollingspider RollingSpider#setup +1ms

/Users/smistry/git/rolling-spider/lib/drone.js:216
  this.peripheral.discoverAllServicesAndCharacteristics(function (error, servi
                  ^
TypeError: Cannot call method 'discoverAllServicesAndCharacteristics' of null
    at Drone.setup (/Users/smistry/git/rolling-spider/lib/drone.js:216:19)
    at /Users/smistry/git/rolling-spider/issue38.js:5:11
    at null.<anonymous> (/Users/smistry/git/rolling-spider/lib/drone.js:188:11)
    at Noble.emit (events.js:95:17)
    at Noble.onStateChange (/Users/smistry/git/rolling-spider/node_modules/noble/lib/noble.js:66:8)
    at emit (events.js:95:17)
    at nobleBindings.startScanning.args.kCBMsgArgOptions (/Users/smistry/git/rolling-spider/node_modules/noble/lib/mac/yosemite.js:70:8)
    at emit (events.js:95:17)
    at nobleBindings.sendCBMsg.sendXpcMessage.kCBMsgId (/Users/smistry/git/rolling-spider/node_modules/noble/lib/mac/yosemite.js:35:8)
    at emit (events.js:95:17)
```

However, the connect callback returns an error, so here's a recommended change:
```javascript
myDrone.connect(function(error) {
  if (error) {
    return console.log(error);
  }

  myDrone.setup(function() {
    myDrone.startPing();
  });
});
```

Which outputs: ```[Error: Error with Bluetooth Adapter, please retry]```

hamzazaidi, 2015-06-06:
I am running Ubuntu on VM, trying to connect the rolling spider and getting this error

hzaidi@ubuntu:~/spiderDrone$ nodejs index.js
noble warning: adapter does not support Bluetooth Low Energy (BLE, Bluetooth Smart).
               Try to run with environment variable:
               [sudo] NOBLE_HCI_DEVICE_ID=x node ...

/home/hzaidi/spiderDrone/node_modules/rolling-spider/lib/drone.js:216
  this.peripheral.discoverAllServicesAndCharacteristics(function (error, servi
                  ^
TypeError: Cannot call method 'discoverAllServicesAndCharacteristics' of null
    at Drone.setup (/home/hzaidi/spiderDrone/node_modules/rolling-spider/lib/drone.js:216:19)
    at /home/hzaidi/spiderDrone/index.js:7:13
    at null.<anonymous> (/home/hzaidi/spiderDrone/node_modules/rolling-spider/lib/drone.js:188:11)
    at Noble.EventEmitter.emit (events.js:95:17)
    at Noble.onStateChange (/home/hzaidi/spiderDrone/node_modules/rolling-spider/node_modules/noble/lib/noble.js:66:8)
    at EventEmitter.emit (events.js:95:17)
    at nobleBindings.onStateChange (/home/hzaidi/spiderDrone/node_modules/rolling-spider/node_modules/noble/lib/linux/bindings.js:32:8)
    at EventEmitter.emit (events.js:95:17)
    at HciBle.onStdoutData (/home/hzaidi/spiderDrone/node_modules/rolling-spider/node_modules/noble/lib/linux/hci-ble.js:62:12)
    at Socket.EventEmitter.emit (events.js:95:17)
hzaidi@ubuntu:~/spiderDrone$ ^C
hzaidi@ubuntu:~/spiderDrone$ ^C
hzaidi@ubuntu:~/spiderDrone$ nodejs index.js
noble warning: adapter does not support Bluetooth Low Energy (BLE, Bluetooth Smart).
               Try to run with environment variable:
               [sudo] NOBLE_HCI_DEVICE_ID=x node ...
[Error: Error with Bluetooth Adapter, please retry]

sandeepmistry, 2015-06-06:
@hamzazaidi what USB Bluetooth adapter are you using? Does the VM have access to it?

voodootikigod, 2015-06-08:
I am thinking of handling this more elegantly than just blowing up, any reason not to ( @sandeepmistry I agree with your implications, most times this is due to either the BT being off or not ready yet )

sandeepmistry, 2015-06-09:
@voodootikigod there's also the scenario where the adapter is not supported (non-bluetooth 4.0).

Resseguie, 2015-06-09:
Good grief. That was it. I just didn't have bluetooth turned on. I had checked that my MBP had 4.0, etc but apparently had just never turned it on.

But agree with @voodootikigod that an elegant error message would have been much more obvious than the one in this issue title.

And perhaps one of the examples in README should include use of the error parameter so it's obvious that's an option?

thanks! I'm at least flying now. Now on to more awesome...

hamzazaidi, 2015-06-09:
@sandeepmistry buildin adapter in the DELL laptop according to the device manage is Broadcom DW1560 Bluetooth 4.0 LE

kevinold, 2015-06-13:
I'm seeing this issue as well on 1.2.0 and latest master (1.2.1).  Was working fine late May.  I've verified bluetooth is turned on, but I do receive the same error when trying to take off for instance with keyboard.js.

The solution turned out to be restarting my Mac.

Resseguie, 2015-06-14:
Just cross referencing that at least part of this issue (better error message) covered by 2ce7d0fda45a29324ec2869f36c39bf95fcb9049

voodootikigod, 2015-06-23:
I am going to close this unless there is a great reason to keep open - i added a lot more error support and output around this. Please letme know if I need to re-open

closed 2015-06-23 by voodootikigod