voodootikigod/node-rolling-spider #40: How to connect several drones

shiva02, 2015-06-04:
Hi everyone, i try to take off 2 rolling spider at the same time by creating two instances and by specifying their uuid but the program sequence is random, sometimes drone 1 and sometimes drone 2, the other times both. Can you help me, can i have a simple example?
Thanks, best regards

Seb V

voodootikigod, 2015-06-04:
It should never do both.. That's an odd case swarm (more than 1) isn't yet
supported. Variable single is supported

shiva02, 2015-06-04:
Ah pity, is planned one day perhaps? I'll keep you posted on tests I plan to do.

voodootikigod, 2015-06-04:
Yup. And soon as well.

-- 
*Chris Williams*

@voodootikigod <http://twitter.com/voodootikigod> | GitHub
<http://github.com/voodootikigod>

*The things I make that you should check out:*
SaferAging <http://www.saferaging.com/> | JSConf <http://jsconf.com/> |
RobotsConf <http://robotsconf.com/> | RobotsWeekly
<http://robotsweekly.com/>

Help me end the negativity on the internet, share this
<http://jsconf.eu/2011/an_end_to_negativity.html>.

voodootikigod, 2015-06-08:
https://github.com/FluffyJack/node-rolling-spider/tree/swarm

Readme coming soon but short is in 

https://github.com/FluffyJack/node-rolling-spider/blob/swarm/eg/swarm.js

voodootikigod, 2015-06-08:
Duplicate of #31

closed 2015-06-08 by voodootikigod

shiva02, 2015-06-08:
Thank you very much, I had not seen this post.
By