voodootikigod/node-rolling-spider #41: How to use this.logger instead of console.log

shiva02, 2015-06-05:
Hello,
I try to use the "logger" option in the settings like in a doc constructor in drone.js file:
/**
 * Constructs a new RollingSpider
 * @param {Object} options to construct the drone with:
 *  - {String} uuid to connect to. If this is omitted then it will connect to the first device starting with 'RS_' as the local name.
 *  - logger function to call if/when errors occur. If omitted then uses console#log
 * @constructor
 */

But I can not.
I try : 
var mydrone = new RollingSpider({uuid : "RS_R112233"}, logger);
or
var mydrone = new RollingSpider({uuid : "RS_R112233", logger = true});

Do you have a solution maybe?
Best regard

voodootikigod, 2015-06-08:
Here is an example:

```
var mydrone = new RollingSpider({uuid: 'RS_R112233', logger: console.log});
```

closed 2015-06-08 by voodootikigod

shiva02, 2015-06-08:
Waoou excuse me, I had read "If omitted then uses console#log", I thought not having to put something in parameter..
Thanks you

voodootikigod, 2015-06-08:
Agh will fix the docs! Thanks!

-- 
*Chris Williams*

@voodootikigod <http://twitter.com/voodootikigod> | GitHub
<http://github.com/voodootikigod>

*The things I make that you should check out:*
SaferAging <http://www.saferaging.com/> | JSConf <http://jsconf.com/> |
RobotsConf <http://robotsconf.com/> | RobotsWeekly
<http://robotsweekly.com/>

Help me end the negativity on the internet, share this
<http://jsconf.eu/2011/an_end_to_negativity.html>.