voodootikigod/node-rolling-spider #43: Error when trying to run flip commands

kevinold, 2015-06-13:
On latest master (1.2.1) I receive the following error when issuing a flip command from eg/keyboard.js. 

Below is the output:

Configured for Rolling Spider!  RS_R107537
ready for flight
takeoff
takeoff
takeoff

/Users/kold/github_clones/node-rolling-spider/node_modules/noble/lib/characteristic.js:60
      callback(null);
      ^
TypeError: object is not a function
    at Characteristic.<anonymous> (/Users/kold/github_clones/node-rolling-spider/node_modules/noble/lib/characteristic.js:60:7)
    at Characteristic.g (events.js:180:16)
    at Characteristic.emit (events.js:92:17)
    at Noble.onWrite (/Users/kold/github_clones/node-rolling-spider/node_modules/noble/lib/noble.js:279:20)
    at emit (events.js:106:17)
    at nobleBindings.write (/Users/kold/github_clones/node-rolling-spider/node_modules/noble/lib/mac/yosemite.js:550:10)
    at Noble.write (/Users/kold/github_clones/node-rolling-spider/node_modules/noble/lib/noble.js:272:19)
    at Characteristic.write (/Users/kold/github_clones/node-rolling-spider/node_modules/noble/lib/characteristic.js:64:15)
    at Drone.writeTo (/Users/kold/github_clones/node-rolling-spider/lib/drone.js:342:49)
    at frontFlip (/Users/kold/github_clones/node-rolling-spider/lib/drone.js:605:10)

voodootikigod, 2015-06-14:
This appears to be a noble issue more than rolling spider -- its calling the callback before checking if the provided callback is valid.

kevinold, 2015-06-14:
Indeed. I filed here for context and meant to circle back yesterday to clarify what you've mentioned and cc @sandeepmistry

voodootikigod, 2015-06-14:
Can you try with 1.3.0 I think I found the bug in my side... possibly or another bug that would have come out later ;)

kevinold, 2015-06-14:
Yeah that patch looks like it will work. I was headed in that direction yesterday but was managing a frustrated 4 year old upset because it didn't flip ;) Will give it a try this afternoon.

voodootikigod, 2015-06-14:
also if you have a logitech f310 hanging around check out `eg/gamepad.js` ;)

*Chris Williams*

@voodootikigod <http://twitter.com/voodootikigod> | GitHub
<http://github.com/voodootikigod>

*The things I make that you should check out:*
SaferAging <http://www.saferaging.com/> | JSConf <http://jsconf.com/> |
RobotsConf <http://robotsconf.com/> | RobotsWeekly
<http://robotsweekly.com/>

Help me end the negativity on the internet, share this
<http://jsconf.eu/2011/an_end_to_negativity.html>.

commit ae8b37d7 (2015-06-15): Merge pull request #44 from kevinold/fix-flip

Check if callback is a function before passing or calling.  Fixes #43
commit ff0f8f51 (2015-06-15): Check if callback is a function before passing or calling.  Fixes #43
closed 2015-06-15 by voodootikigod