voodootikigod/node-rolling-spider #46: Swarm with different trajectory

shiva02, 2015-06-20:
Hello everyone.
Is it possible to make a different trajectory for each drones of swarm ?
May be using an index element of the swarm like this : 
   swarm.membership[0].takeOff()
If you have an idea?

voodootikigod, 2015-06-22:
Use `swarm.at('RS_W123456').forward()`

voodootikigod, 2015-06-22:
https://github.com/voodootikigod/node-rolling-spider#swarmatid-callback

closed 2015-06-22 by voodootikigod

shiva02, 2015-06-22:
How thanks...
I am beginner and I had not seen the new doc...
Thanks a lot

voodootikigod, 2015-06-22:
No worries, just got it up actually because you filed the ticket so thank
you for the nudge ;) cheers and best of flying

*Chris Williams*

@voodootikigod <http://twitter.com/voodootikigod> | GitHub
<http://github.com/voodootikigod>

*The things I make that you should check out:*
SaferAging <http://www.saferaging.com/> | JSConf <http://jsconf.com/> |
RobotsConf <http://robotsconf.com/> | RobotsWeekly
<http://robotsweekly.com/>

Help me end the negativity on the internet, share this
<http://jsconf.eu/2011/an_end_to_negativity.html>.