voodootikigod/node-rolling-spider #49: Error after landing swarm and solved (perhaps)

shiva02, 2015-06-24:
Hi,

When i use swarm.js in eg directory an when i press "q" for landing and error appears "this.members = []"
I change line 220 in swarm.js in lib directory : 
this.members = []  ---> changed to this.membership = [];
And it works...

voodootikigod, 2015-06-24:
Can you submit a pull request please?

shiva02, 2015-06-24:
Yes of course, but I do not know too Github, how do I do once he asks me to choose the branch?

voodootikigod, 2015-06-24:
https://help.github.com/articles/creating-a-pull-request/

Use this is as a guide, ping if you need more help

Resseguie, 2015-06-24:
@shiva02 for a simple one line change like that, you can also go to the file directly on github (https://github.com/voodootikigod/node-rolling-spider/blob/master/lib/swarm.js), click the pencil icon near the top-right of the code, and make the change there. When you click "propose file change" when you're done, it creates a branch for you and initiates a PR (pull request) for you. I usually avoid that for large changes, but for simple ones like this, it  can be a quick solution.

shiva02, 2015-06-24:
@Resseguie
Great,
I do this now, thanks Resseguie

Shiva02

closed 2015-06-24 by shiva02