voodootikigod/node-rolling-spider #50: Not all membership take off

shiva02, 2015-06-24:
Re,

All the member does not take off.

This is my swarm var :
"var swarm = new Swarm({membership:'RS_A112233,RS_B112233,RS_C112233',timeout: 10}); 
I can not control all the drones, only one of them takes off and not necessarily the same at each execution. Can you help me again ?

Thanks a lot

Shiva02

voodootikigod, 2015-06-24:
Hey @shiva02 

Can you run again with 

`DEBUG=rollingspider node eg/swarm.js`

Then provide the output here. Your issue with non-members could be due to the timeout (10s) not allowing all of them to register/setup/join.

Also I have never seen a local name with RS_A or RS_C ... like ever... can you verify thats the correct names by running 

`node eg/discover.js` 

and pasting the output in the comment as well!

shiva02, 2015-06-24:
Yep,
RS_A B C, it's just an exemple...you are right I have RS_B or R.

For the deub how to provide the output ?I'm sorry, I'm a beginner hehe

voodootikigod, 2015-06-24:
No worries about being a beginner (aren't we all anyways?) happy to help out!

Best way to provide long files of data is to create a [new Gist](https://gist.github.com/) and paste it in there. You can paste in multiple files to a single "gist". Once you are done pasting in, just copy the URL and paste it in here!

Cheers

shiva02, 2015-06-24:
Thank you for your response, I try to do a little choreography with drones..
Degug result : https://gist.github.com/a70fd4b7af83229f8a9f.git

I hope it's ok for you

shiva02, 2015-06-24:
For precision, I use a temporal module for execute swarm action with time sequence (like eg/index.js)
New link for debug file if first not work : 
https://gist.github.com/shiva02/a70fd4b7af83229f8a9f

shiva02, 2015-06-25:
Hello, sorry back to you but I would really like to understand what is happening. Perhaps can be should i do save the drones in blutooth with linux command "hci" specific or simply click on "to trust" for each of the drones in blutooth interface linux?
Best regard

Shiva02

johnelliott, 2015-07-01:
@voodootikigod great to see swarm is working :)

aamodh, 2015-11-17:
shiva02 did you figure out a solution? I am facing a similar problem with 3 drones. Not all of them are connecting even though it recognises their existence. I played around with the timeout values but still its pretty unpredictable.

shiva02, 2015-11-18:
Good evening, I'm travelling, I do not have my development computer. I try to
recover my code and you answers. I have managed to make six drones take off
at the same time but it is not "super" reliable. This end of week i gives
you information..

Best regards

aamodh, 2015-11-18:
Oh thanks for your reply. That sounds promising. Not a problem, I'll wait for your updates :).

anandx, 2016-05-23:
Hi there, I am having the same problem reliably connecting to all 3 of my drones.  80% of the time it will connect to only 1.  Very rarely 2 but never all 3. My logs look the same as [shiva's above](https://gist.github.com/shiva02/a70fd4b7af83229f8a9f).  

Any thoughts whats up?  I'm sending the UUIDs, and forceConnect does not seem to help either.  All 3 UUIDs do show up in discover.js however.  @voodootikigod Any ideas?

anandx, 2016-05-23:
After digging a little deeper into what's happening, sometimes the drones that don't connect are returning  "undefined is not a member" in the assemble() function.  I've added code to output the peripheral object and you can see that peripheral.advertisement does not contain a localName or manufacturerData field for the "not a member" drone.  

See logger output here, key lines tagged with "$$$$$$$$$$$$":  
https://gist.github.com/anandx/3502dca57f53c42c2ccefe86905263f5

This is especially strange because when I dump the peripheral objects in discover.js they contain well formed advertisement objects for all 3 drones.  What the heck is going on!  Is this problem at the noble level or node-rolling-spider?  

(Note: there is also 1 drone that appears in discover but is not visible at all in the gist output, but thats another problem)

aamodh, 2016-05-23:
Hi Anand
I faced numerous issues as well, I did manage to fly 3 drones at once but
there was considerable lag especially if I was sending different commands
(15-20 times a sec) to each drone. I think its the noble platform itself,
but i am not sure. So now I am connecting each drone to individual
Bluetooth dongle and have different virtual machines running
simultaneously. I am still testing this out, will keep you posted. Please
do let me know if you figure out a reliable way to control multiple drones.

anandx, 2016-05-24:
Hi @aamodh, 
Thanks for your reply!  How did you eventually get the 3 drones running?  Only 2 of 3 connect for me and thats only about 20% of the time. When they do however they move quiet in sync and without lag.

anandx, 2016-05-24:
Further update.  I think the problem may well be in the node-rolling-spider swarm.js code.  

I modified eg/discover.js to include .connectPeripheral() code from swarm.js.  [code here](https://gist.github.com/anandx/d8e90a4d1ec401734f3ced138cc4dd34).  And when I do it is able to connect & setup to all 3 drones 100% of the time!  [logs](https://gist.github.com/anandx/14f5289c2cc189d87d7b6b8ab74ca4af).  Whereas I can't even reliably get 3 peripheral.advertisement.manufacturerData fields in swarm.js

So maybe there is some funkiness in how swarm.js is connecting?  The code does seem to be pretty involved, having some trouble wrapping my head around it.

anandx, 2016-06-03:
Btw, i got swarms to reliably work communicating with each drone direclty.  Not sure what is going on in swarm.js as the code is quite involved but this simple approach works for me.  Hope they can fix swarm.js!  

[swarm-lite.js](https://gist.github.com/anandx/5e21313a50b81c2ab6bab45172baa539)

lynnaloo, 2017-04-05:
@anandx does the current `swarm.js` work for you?

mztor, 2018-05-26:
Your fix was really helpful @anandx. The only other thing we had to do was comment out line 223 in drone.js:
`this.ble.stopScanning();`
This was a line in the connectPeripheral function. Not sure what it does, but hopefully, commenting it out won't be too big a deal.
Thank you so much!