voodootikigod/node-rolling-spider #60: Not in the air?

picsoung, 2015-07-26:
Hey,

I am playing around the example script, all good, takes off, goes forward, and lands.

I tried to add a `frontflip` command between existing tasks but in the console I see `
Calling RollingSpider#frontFlip when it's not in the air isn't going to do anything`

while my robot is in the air, and flying...
what could be wrong?

in the same time: it looks like I have to reset it everytime I want him to execute a program, even after calling `disconnect` at the end of the program.

Thanks

markwest1972, 2015-08-02:
Hello!

I am also experiencing this problem.  As far as I can tell this is due to the *drone.status.flying* value being set to false once the drone is in the air.  

The value is being changed at line 279 in the *drone.js* file, due to the value of data[6] being either "undefined" or "0".

Anyone have any ideas?  Surely this is a bug and not a feature?

Mark

anandx, 2016-05-02:
Is this issue fixed for you in the latest commit?  https://github.com/voodootikigod/node-rolling-spider/pull/81/files

When I run the basic [take off/move forward/land code](https://github.com/voodootikigod/node-rolling-spider#taking-off-moving-and-landing) the drone still thinks its not flying and I get "when it's not in the air isn't going to do anything" in response to the forward & land commands and the script dies with the drone hovering in the air. 

One time it died while the drone was driving forward and didn't stop until it crashed into the wall, any idea what might be up?  Why does the drone think the state is not flying when it is?

deadprogram, 2016-05-02:
Hi @anandx I tested it on OS X and Linux by running `eg/index.js` program.

```
$ node eg/index.js 
Connected to drone RS_R121231
Getting ready for takeOff!
Going forward
Going up
Going down
Going left
Going right
OMG Flip!
Time to land
```

Are you sure you are running the code from the hybridgroup fork? This PR has not been merged into the main `voodootikigod/master` as of yet.

anandx, 2016-05-04:
I'm pretty sure.  I pulled directly from [here](https://github.com/voodootikigod/node-rolling-spider/pull/81).  I will try again and test with my 3 drones however I am traveling for the next 10 days without my drones :(  Will update when I return.

deadprogram, 2016-05-07:
We have been using our fork very successfully, which is what this PR is based on. You can try it from here https://github.com/hybridgroup/node-rolling-spider

Also, you should probably make sure you are running the latest drone firmware. The only way to update the drone firmware currently is to use the freeflight mobile app..

anandx, 2016-05-18:
Ok I discovered the problem.  It's basically in two differences between the index.js code and the [take off/move forward/land code](https://github.com/voodootikigod/node-rolling-spider#taking-off-moving-and-landing) posted in the [Readme.md](https://github.com/voodootikigod/node-rolling-spider).  The code in the readme causes problems and should be updated to be like index.js as follows:

1)  the init steps should be as follows, otherwise for some reason the drone loses communication with the PC after the first command
```var RollingSpider = require('../');
var temporal = require('temporal');
var rollingSpider = new RollingSpider(process.env.UUID);
```

2) the .forward() call should include a fixed number of steps otherwise it just keeps flying. 
`rollingSpider.forward({steps: 12});`

Hope someone can update the readme, not quite sure how that works.

lynnaloo, 2017-04-08:
I updated the README.md. Let me know if I can close this.

closed 2018-01-08 by lynnaloo