voodootikigod/node-rolling-spider #66: New Parrot Airborne compatibility

jtbonhomme, 2015-11-11:
Hi,

Do you think this lib could work with the new Parrot's Airborne drones ?

Best regards,
Jean-Thierry

AlbertoHdezCerezo, 2015-11-22:
Hi jtbonhomme,

Days ago I had the same doubt, but consulting on the internet I saw that the Airborne is a modification of the Spider, so, the answer should be yes (they are the same device with different design).

jtbonhomme, 2015-11-23:
Hi,

Thank you for your message. I bougth an Airborne minidrone and tested with a recent mac book pro laptop (with BLE) and it's Ok !

Regards,
Jean-Thierry

jtbonhomme, 2016-01-21:
Thanks for your message, I had the same feedback from a Parrot's tech
leader, he kindly answered to me. I did not realize that my macbook was too
old and did not have BLE :)

closed 2017-04-05 by lynnaloo