voodootikigod/node-rolling-spider #67: Usign cylon-rolling-spider with Meteor

AlbertoHdezCerezo, 2015-11-23:
Hi!

Recently I have been starting to develop APPs based on Meteor, and I would like to develop an APP that allow users to drive a rolling spider using your package.

Despite Meteor has not implemented a "require" method to load NPM packages, I was capable of loading packages using meteorhacks:npm and cosmos:browserify (using chalk as an example). However, when I am trying to do the same with your package an error is triggered.

Do I have to set any additional dependence to the package or do any additional step to make it work? Right know I am a bit lost U.U

roeku, 2015-12-07:
Hey, I'm having trouble getting it working, did you have any luck figuring it out?

AlbertoHdezCerezo, 2015-12-07:
Nope, I am looking for a different solution but right now I see impossible to make it works with Meteor U.U