voodootikigod/node-rolling-spider #69: Can get info about drone, cannot 'connect' to drone?

dronestarter, 2015-12-08:
Hi, first post, I hope I don't make too many mistakes.

I've installed your great lib, but can't seem to get control of my rolling spider. For example, 

$ node node_modules/rolling-spider/eg/keyboard.js

just hangs. With debug output I see:

```  rollingspider RollingSpider#connect +0ms
  rollingspider RollingSpider.on(stateChange) +3ms
  rollingspider RollingSpider#poweredOn +9ms
  rollingspider RollingSpider.on(discover) +38ms
  rollingspider Peripheral { _noble: Noble { state: 'poweredOn', _bindings: EventEmitter { _state: 'poweredOn', _addresses: [Object], _addresseTypes: [Object], _connectable: [Object], _pendingConnection: false, _connectionQueue: [], _handles: {}, _gatts: {}, _aclStreams: {}, _hci: [Object], _gap: [Object], _events: [Object], _eventsCount: 20, onSigIntBinded: [Function: bound ], _scanServiceUuids: [] }, _peripherals: { e0145e22[redacted]: [Circular] }, _services: { e0145e22[redacted]: {} }, _characteristics: { e0145e22[redacted]: {} }, _descriptors: { e0145e22[redacted]: {} }, _discoveredPeripheralUUids: [ 'e0145e22[redacted]' ], _events: { warning: [Object], discover: [Function: bound ], stateChange: [Function: bound ] }, _eventsCount: 3, _allowDuplicates: undefined }, id: 'e0145e22[redacted]', uuid: 'e0145e22[redacted]', address: 'e0:14:5e:22:3d:6a', addressType: 'random', connectable: true, advertisement: { localName: 'RS_W023[redacted]', txPowerLevel: undefined, manufacturerData: <Buffer 43 00 cf 19 00 09 01 00>, serviceData: [], serviceUuids: [] }, rssi: -44, services: null, state: 'disconnected' } +0ms
  rollingspider RS_W023[redacted] +19ms
  rollingspider Fuzzy match found: RS_W023[redacted] <e0145e22[redacted]> +0ms
  rollingspider RollingSpider#down +4s
  rollingspider RollingSpider#down when it's not in the air isn't going to do anything +1ms

```

Sorry if the redacting makes the output useless, I'm not sure how much of that info you need, nor which is actually unique.

Thanks for any help you might be able to offer me.

dronestarter, 2015-12-16:
Sorry to spam, but does anyone know what might be causing this? I was really looking forward to scripting this drone.

Thanks!

lynnaloo, 2017-04-05:
@dronestarter is this still an issue?

closed 2018-01-08 by lynnaloo