voodootikigod/node-rolling-spider #70: Support parrot cargo ?

marianodominguez, 2016-01-04:
According to doc, the difference with rolling spider is the lego pieces, wonder if software is too different.

I was able to connect to device, but it seems it does not hold naming convention:

`
  rollingspider Peripheral { _noble: Noble { state: 'poweredOn', _bindings: EventEmitter { _state: 'poweredOn', _addresses: [Object], _addresseTypes: [Object], _connectable: [Object], _pendingConnection: false, _connectionQueue: [], _handles: {}, _gatts: {}, _aclStreams: {}, _hci: [Object], _gap: [Object], _events: [Object], _eventsCount: 20, onSigIntBinded: [Function: bound ], _scanServiceUuids: [] }, _peripherals: { '68644b1f30cc': [Object], e014e7343da3: [Circular] }, _services: { '68644b1f30cc': {}, e014e7343da3: {} }, _characteristics: { '68644b1f30cc': {}, e014e7343da3: {} }, _descriptors: { '68644b1f30cc': {}, e014e7343da3: {} }, _discoveredPeripheralUUids: [ '68644b1f30cc', 'e014e7343da3' ], _events: { warning: [Object], discover: [Function: bound ], stateChange: [Function: bound ] }, _eventsCount: 3, _allowDuplicates: undefined }, id: 'e014e7343da3', uuid: 'e014e7343da3', address: 'e0:14:e7:34:3d:a3', addressType: 'random', connectable: true, advertisement: { localName: 'Travis_050419', txPowerLevel: undefined, manufacturerData: <Buffer 43 00 cf 19 09 09 01 00>, serviceData: [], serviceUuids: [] }, rssi: -59, services: null, state: 'disconnected' } +0ms
  rollingspider Travis_050419 +3ms
`

sandeepmistry, 2016-01-05:
@marianodominguez the manufacturer data does match the check here: 
https://github.com/voodootikigod/node-rolling-spider/blob/master/lib/drone.js#L94

Once you connect can you take off? See [keyboard.js](https://github.com/voodootikigod/node-rolling-spider/blob/master/eg/keyboard.js) example.

If that works, a similar PR to #64 could be added to support it.

marianodominguez, 2016-01-05:
Hello, I think the problem is the local name not staring with "RS",  so it is not recognized as, connects just ignore it forever.

I'll test removing the restriction on name.

virgilvox, 2016-01-05:
I just put up a pull request for commit #71 

@marianodominguez your suggestion worked :+1:

marianodominguez, 2016-01-05:
@virgilvox added my model to pull request #72 ... my drone is taking off :)

sandeepmistry, 2016-01-05:
@marianodominguez @virgilvox awesome, great news!

It would be good to also add checks to match on the manufacturer data value, as the local name can be changed via the Parrot mobile app. (See #26)

marianodominguez, 2016-01-05:
@sandeepmistry manufacturer id does not match, either : for my Travis is 

`manufacturerData: <Buffer 43 00 cf 19 09 09 01 00>`
`manufacturer.toString('hex')`

'4300cf1909090100' 

yours is 

'4300cf1900090100'

@virgilvox which mfr does your mars retrieve?

virgilvox, 2016-01-06:
@marianodominguez for Mars_ its 4300cf1909090100

Frostlock, 2016-01-27:
I stumbled on the same thing for the Airborne night swat drone.
Also I renamed my drone earlier through freeflight3 on ipad. So I had to get rid of the name check.

lynnaloo, 2017-04-05:
There was a bug in that manufacturer check that's fixed in 1.6.0. This should work for you now.

closed 2017-04-05 by lynnaloo