voodootikigod/node-rolling-spider #74: Discover works fine, but unable to run index.js with success

fucci, 2016-01-14:
Colleague is running successful with Mac, I am running Ubuntu 14.04 with noble, temporal, rolling-spider, and bluetooth-hci-socket node modules. 

Discover showed my minidrone
1: RS_B075911 (e014b3783d48), RSSI -54

When I run index.js, just hangs

Any advice is appreciated, 
also any pointers on how to get debugging or logging on also appreciated.

Apologies if I sound dumb..... still getting up to speed with node

Thanks for this great piece of code/functionality

roeku, 2016-01-14:
I had the same problem

https://github.com/voodootikigod/node-rolling-spider/issues/68

fucci, 2016-01-14:
Thanks roeku, you mentioned, the following - so the BLE stops scanning, and it can then set up
this.ble.stopScanning();
- is this correct

Also, there are 2 more things I should mention
A. when I say the index.js hangs, I get the following coming up, which I think it means it stops
1: RS_B075911 (e014b3783d48), RSSI -36
B. assuming it does not matter where I run the index.js files

appreciate your help

fucci, 2016-01-14:
Thanks roeku, you mentioned, the following - so the BLE stops scanning, and it can then set up
this.ble.stopScanning();

    is this correct

Also, there are 2 more things I should mention
A. when I say the index.js hangs, I get the following coming up, which I think it means it stops
1: RS_B075911 (e014b3783d48), RSSI -36
B. assuming it does not matter where I run the index.js files

appreciate your help

roeku, 2016-01-14:
Go into lib/drone.js and add this line to 201
 ```this.ble.stopScanning(); ```
under 
      ``` this.peripheral = peripheral;```

And then execute the index.js or another file in the eg folder. As far as I know this should resolve your issue.

fucci, 2016-01-15:
yippee !!!!

that worked - many thanks

I am still a little confused, why I had to include that one line of code in
my version on ubuntu
- when my colleague on Mac did not

regardless - I am very happy
- Thank You

beverts312, 2016-03-28:
Thanks @roeku !

barackd222, 2016-08-24:
For us this solved the problem, but not after having to re-connect
Thank you

Thyxx, 2017-03-19:
Hi, 
Same problem here:
`noble warning: unknown peripheral eXXXXXXXXXXX`

Discover.js works fine but I can't run index.js
Already add `this.ble.stopScanning();` but there is no difference.

Drone: Rolling-Spider
Distrib: Debian 8

Thanks in advance.

lynnaloo, 2017-04-08:
Is this still an issue?

Thyxx, 2017-04-10:
Problem solved...
I edited the message.