voodootikigod/node-rolling-spider #78: Docs for Bluetooth API?

poshaughnessy, 2016-01-30:
Hi,

I'm hoping to use this library as the basis for a Web Bluetooth demo. In order to do that, I'm hoping to better understand the underlying Bluetooth API - how to construct the right commands for the drone.

Please may I ask: were there any particular docs that you used to figure out what exactly you needed to write to the characteristics, which services and characteristics you needed, etc?

Thanks!

naus3a, 2016-05-16:
Here's the protocol:
https://github.com/Parrot-Developers/libARCommands/tree/master/Xml

and here's how to use it:
http://forum.developer.parrot.com/t/bluetooth-attribute-definitions-for-roll-pitch-yaw-rotation-and-altitude/105/3

poshaughnessy, 2016-08-23:
Thanks @naus3a and sorry for taking months to reply! The protocol link now gives a 404 - there doesn't seem to be an Xml directory anymore? I've been trying to figure out how to see their docs. I see they have a bunch of Latex (bit of a pain!) files here: https://github.com/Parrot-Developers/Docs/tree/master/Protocol/latex But still can't see info on what bytes to send. Please might you know the latest link to find that?

naus3a, 2016-08-24:
yes @poshaughnessy , they removed it; I'm attaching a copy of the old files.

[xml.zip](https://github.com/voodootikigod/node-rolling-spider/files/434600/xml.zip)

poshaughnessy, 2016-08-24:
Great, thanks @naus3a, I will take a look!

closed 2016-08-24 by poshaughnessy

cvuorinen, 2018-03-20:
In case anyone else is looking for this info, the xml files can be found in the linked repo from some earlier commits in the history, it's version control after all 😉 
E.g. https://github.com/Parrot-Developers/libARCommands/blob/7b5cf7a8009278cf634ee33d5c2ed5dd8e573eb4/Xml/MiniDrone_commands.xml