voodootikigod/node-rolling-spider #82: Free fall take-off

mehmandarov, 2016-05-12:
Do any of you have any idea if it is possible to add a support for free fall take-off? The app has this feature, so it should be also possible to access/implement. Any kind of help would be appreciated.

Thanks!

mehmandarov, 2016-05-15:
I can help with the implementation, but I will need a bit more details about the BT-command that is being sent.

naus3a, 2016-05-16:
Hi,
I'm working on a fork implementing some more commands and I have autotakeoff working; at some point I plan to upload my code, but in the meantime, here's the buffer for the freefall take off:

new Buffer([0x02, ++this.steps.fa0c & 0xFF,
                0x02,
                0x00,
                0x05, 0x00,
                0xFF])

mehmandarov, 2016-05-16:
Hi, naus3a! Thanks a lot!

naus3a, 2016-05-16:
no problem :)

roeku, 2016-05-21:
Anyone able to implement this already?

naus3a, 2016-05-23:
you just need to send the buffer in my last post in the same way the other BT messages are sent :)

lynnaloo, 2017-04-08:
@naus3a if you've already got this working, please submit a PR!

naus3a, 2017-04-10:
I have a fork with some project specific modifications; I'll extract the generic parts and make a PR out of them