voodootikigod/node-rolling-spider #84: "You must have bluetooth enabled and be connected to a drone" error

remzi-e, 2017-03-27:
Not sure if this repo is active but giving this a try:
I'm running this in Ubuntu 16.04 64.
Installed all pre-reqs.
connected ubuntu to the Maclane drone
then ran the `discover.js` with sudo in a terminal
got `noble: unknown peripheral <addr> connected` 
immediately after got `noble: unknown peripheral <addr> disconnected`
then i ran `sudo node reconnect.js` in a different terminal (`reconnect.js` is a bit modified to just do the take off and landing)

i get the following error:
```
Prepare for take off!  Maclan_033049
/home/remzi/Desktop/workspace/node-rolling-spider/lib/drone.js:342
      throw e;
      ^

Error: You must have bluetooth enabled and be connected to a drone before executing a command. Please ensure Bluetooth is enabled on your machine and you are connected.
    at Drone.writeTo (/home/remzi/Desktop/workspace/node-rolling-spider/lib/drone.js:338:13)
    at Drone.flatTrim (/home/remzi/Desktop/workspace/node-rolling-spider/lib/drone.js:631:8)
    at /home/remzi/Desktop/workspace/node-rolling-spider/eg/reconnect.js:17:9
    at Drone.<anonymous> (/home/remzi/Desktop/workspace/node-rolling-spider/lib/drone.js:144:11)
    at Peripheral.<anonymous> (/home/remzi/Desktop/workspace/node-rolling-spider/node_modules/noble/lib/peripheral.js:38:7)
    at Peripheral.g (events.js:291:16)
    at emitOne (events.js:96:13)
    at Peripheral.emit (events.js:188:7)
    at Noble.onConnect (/home/remzi/Desktop/workspace/node-rolling-spider/node_modules/noble/lib/noble.js:151:16)
    at emitTwo (events.js:106:13)
```

Am i missing something?

lynnaloo, 2017-04-05:
@remzi-e is this still an issue?

remzi-e, 2017-04-05:
i still get the `unknown peripheral <addr> connected` error when running the discover.js example
don't get the error though.
should we close this and open a different issue?

lynnaloo, 2017-04-05:
Which Parrot drone is this? That `unknown` error is likely because the `Maclan_` prefix isn't in the list of drones that are supported. I can add that and try to provide a more information in the case of unknown drones.

remzi-e, 2017-04-05:
i get this with both Airborne Night Maclane and the Mambo models.

remzi-e, 2017-04-05:
should this issue be renamed?

lynnaloo, 2017-04-05:
Did you already run this:

`sudo apt-get install bluetooth bluez libbluetooth-dev libudev-dev`

remzi-e, 2017-04-05:
of course. i can get the drones moving (1 by 1) but the `discover.js` script throws that `unknown` output everytime i run it after connecting to the drone from the OS.

lynnaloo, 2017-04-05:
Are you pairing the drone with Bluetooth through Ubuntu before running `discover.js`? If that's the case, then you only want to do it through this application since `noble` does the pairing.

remzi-e, 2017-04-05:
yeah, i thought that was the normal workflow.
if i don't do that before, there is not output from `discover.js`.
`index.js` is successful in connecting and doing what it does, though.

lynnaloo, 2017-04-05:
Can you console.log `peripheral` here: https://github.com/voodootikigod/node-rolling-spider/blob/master/eg/discover.js#L17? I suspect it's not making it past that conditional check. 

Also, make sure you're using version `1.7.0`. We just published that today.

lynnaloo, 2017-04-08:
@remzi-e can we close this one?

remzi-e, 2017-04-08:
Yes.

closed 2017-04-08 by remzi-e