voodootikigod/node-rolling-spider #97: `this` scoping issues with `setTimeout`

rafkhan, 2017-07-02:
It appears as though the line `setTimeout(rollingSpider.land, 5000);` fails with an undefined error because `setTimeout` is setting `this` to the global scope. I'd be willing to contribute a fix if it's not being worked on. ([Solution here](https://stackoverflow.com/questions/2130241/pass-correct-this-context-to-settimeout-callback))

```
// WORKS
setTimeout(function() { rollingSpider.land(); }, 5000);

// BREAKS 
setTimeout(rollingSpider.land, 5000);

...

/Users/rafy/Developer/drone-js-1/node_modules/rolling-spider/lib/drone.js:584
  this.logger('RollingSpider#land');
       ^

TypeError: this.logger is not a function
    at Timeout.land [as _onTimeout] (/Users/rafy/Developer/drone-js-1/node_modules/rolling-spider/lib/drone.js:584:8)
    at ontimeout (timers.js:365:14)
    at tryOnTimeout (timers.js:237:5)
    at Timer.listOnTimeout (timers.js:207:5)
```

lynnaloo, 2017-09-22:
We love pull requests!

rwaldron, 2017-09-22:
> `setTimeout(rollingSpider.land, 5000);`

I can't actually find this code anywhere in the project 0_o

---------------------------------

The `land()` method of a `RollingSpider` instance is `this` sensitive and cannot be called as a an argument to `setTimeout` without having its context explicitly bound. Two solutions, which require no library code or example program changes: 

```js
let land = rollingSpider.land.bind(rollingSpider);
setTimeout(land, 5000);
```

```js
setTimeout(() => rollingSpider.land(), 5000);
```

The second is likely the best.

rafkhan, 2017-09-22:
@rwaldron I ended up using the first pattern back when I opened this issue. I wanted to reuse the reference to land, so the first was definitely easier. I took a stab at changing the library so I wouldn't have to bind every function I wanted to use. When I'm at my personal computer again, I can try and dig it up, dust it off and send a PR :)