voodootikigod/node-rolling-spider #103: Bluetooth issue on High Sierra preventing takeoff after landing.

pchinjr, 2018-01-17:
Hi hi hi! 
I'm using MacOS High Sierra 10.13.2
Noble does not work currently from the master branch for High Sierra. To work around this, there is a pull request open with High Sierra support:  https://github.com/sandeepmistry/noble/pull/689/
This is working ok to control the drone and take off, however, after landing, it will not take off again. It seems that the Noble High Sierra pull request does not register notifications correctly. 

I have confirmed that `isNotification` being passed to `.getCharacteristic` is always false on the High Sierra Pull Request. When `isNotification` is always false, `node-rolling-spider` will not emit `flyingStatusChange` on lines 285-291.  This keeps the state of flying to false after it lands and prevents take off. When I comment out lines 287-290, or manually set `this.status.flying = false` in the `land` function, then it will take off and land and take off again. 
https://github.com/voodootikigod/node-rolling-spider/blob/master/lib/drone.js#L285-L291

I don't know enough about how bluetooth works inside the Noble library to fix this registration of notifications. I also don't understand what the `isNotification` check is used for and why I can comment it out and it seems to be ok?

Thanks everyone!

vvo, 2018-01-26:
@pchinjr I have the exact same issue, the lines you outlined are never reached and I have no way to understand why.

Maybe @N-Bz you can give a hand here, or @jacobrosenthal? Thanks!

jacobrosenthal, 2018-01-26:
This is presumably the line you're having problems with.
https://github.com/jacobrosenthal/noble/blob/highsierra/lib/mac/highsierra.js#L522

Id put a console.log(args) here on a high sierra machine https://github.com/jacobrosenthal/noble/blob/highsierra/lib/mac/highsierra.js#L520

and here on a mavericks+ machine https://github.com/jacobrosenthal/noble/blob/highsierra/lib/mac/mavericks.js#L385

and see what apple changed in args.kCBMsgArgIsNotification

vvo, 2018-02-02:
Ok it seems like we need to update the lib but it should work: https://github.com/sandeepmistry/noble#characteristic-1

cc @jacobrosenthal do you confirm? Thanks!

jacobrosenthal, 2018-02-02:
node rolling spider needs to update to noble@^1.9 and not rely on the isnotification flag anymore